chipsalliance/synlig #2066: Import package error

sifferman, 2023-10-12:
Hello. Thanks for this tool! I've been using it for a year now and love it!

## Issue

I found an error with package imports.

I have a package `"my_pkg.sv"`, and two tests `"test1.sv"` and `"test2.sv"`:

```systemverilog
package my_pkg;
    localparam NUM = 1;
endpackage
```
```systemverilog
module test1 import my_pkg::*; (
    input a,
    output b
);
    assign b = (a ^ NUM);
endmodule
```
```systemverilog
module test2 (
    input a,
    output b
);
    import my_pkg::*;
    assign b = (a ^ NUM);
endmodule
```

I would expect that I can include my_pkg.sv first, then testX.sv. (This would match VCS/Verilator/more) However, I found that I must use `-defer` and `-link` to get it to work:

```shell
read_systemverilog my_pkg.sv
read_systemverilog testX.sv # error

read_systemverilog testX.sv # error
read_systemverilog my_pkg.sv

read_systemverilog -defer my_pkg.sv
read_systemverilog -defer testX.sv
read_systemverilog -link # works
```

Error received:

```log
Executing Verilog with UHDM frontend.
ERROR: Error when parsing design. Aborting!
```

Is it possible for this to be changed? Or at least can a warning be issued that package imports only work with `-defer`+`-link`? Thanks!

alaindargelas, 2023-10-12:
@sifferman , simply type:
read_systemverilog my_pkg.sv testX.sv

sifferman, 2023-10-24:
Thanks! That resolves my issues

closed 2023-10-24 by sifferman