chipsalliance/synlig #2069: ERROR: Currently support for dot-access is limited to single range

grotival, 2023-10-13:
I hit this case when parsing some advanced Google SystemVerilog with UHDM + Yosys. Not sure what is the exact repro case. I can look into it more if needed, just thought I would flag it ASAP.

line throwing this error: https://github.com/chipsalliance/yosys-f4pga-plugins/blob/474724189715dd0d4fbb51d2f1cb2fa22f205722/systemverilog-plugin/UhdmAst.cc#L765

FYI: @hzeller @QuantamHD

kgugala, 2023-10-13:
@grotival we moved the plugin code to separate repository `Synlig`. I've moved the issue here, so we can keep track on it

alaindargelas, 2023-10-13:
@grotival all the legal cases in this issue actually work.

The following code as you requested is illegal (No commercial simulator supports it):
```
typedef struct packed {logic [3:0][7:0] k;} my_struct_packed_t;

module single_range_dot_access (
    input  my_struct_packed_t x2,
    output my_struct_packed_t y2
);
  assign y2[1][1] = x2[2][3];
endmodule : single_range_dot_access
```

If you correctly declare x2 and y2, then it works:

```
typedef logic [3:0][7:0] my_struct_packed2_t;

module single_range_dot_access (
    input  my_struct_packed2_t x2,
    output my_struct_packed2_t y2
);
  assign y2[1][1] = x2[2][3];
endmodule : single_range_dot_access
```

Closing this case as it is complete given legal Verilog.

closed 2023-10-13 by alaindargelas