chipsalliance/synlig #2085: "read_systemverilog -lowmem" causes segmentation fault

YikeZhou, 2023-10-20:
## Versions

- Synlig: [Latest release (2023-10-18-bdac131)](https://github.com/chipsalliance/synlig/releases/tag/2023-10-18-bdac131)
- Yosys: `d21c464ae`

## Reproduction Steps

It seems that the problem has nothing to do with the input `.sv` file.
Therefore, I'll use this `top.sv` as an example.
```
module top;
endmodule
```

I learned from `README.md` that:
> arguments to this command should be exactly the same as for Surelog binary.

So I run yosys with this command: (`-lowmem` is an option for Surelog)
```
yosys -p "plugin -i systemverilog; read_systemverilog -lowmem top.sv"
```

This leads to a segmentation fault:
```

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.34+43 (git sha1 d21c464ae, clang 14.0.6 -fPIC -Os)

-- Running command `plugin -i systemverilog; read_systemverilog -lowmem top.sv' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file "/app/slpp_all/surelog.log".
Running: cd "/app"; "/usr/local/bin/yosys" -o "/app" -nostdout -batch /app/slpp_all/mp_parser/pp_batch.txt
/usr/local/bin/yosys: invalid option -- 'n'
Run '/usr/local/bin/yosys -h' for help.
Surelog preproc status: 256
Running: cd "/app"; "/usr/local/bin/yosys" -o "/app" -nostdout -batch /app/slpp_all/mp_preprocess/parser_batch.txt
/usr/local/bin/yosys: invalid option -- 'n'
Run '/usr/local/bin/yosys -h' for help.
Surelog parsing status: 256
Segmentation fault (core dumped)
```

The content in `surelog.log`:
```
********************************************
*  SURELOG SystemVerilog  Compiler/Linter  *
********************************************

Copyright (c) 2017-2023 Alain Dargelas,
http://www.apache.org/licenses/LICENSE-2.0

VERSION: 1.76
BUILT  : Oct 18 2023
DATE   : 2023-10-20.06:27:32
COMMAND: -lowmem top.sv -DYOSYS=1 -DSYNTHESIS=1

[INF:CM0023] Creating log file "/app/slpp_all/surelog.log".
```

tgorochowik, 2023-10-20:
Hi, unfortunately the `-lowmem` option is one of the options that will not work with the plugin directly, as it doesn't work properly when using Surelog API (CC @alaindargelas ).

What the option does it executes Surelog multiple times (by calling the `$0` - the original executable).

To work around this issue you an use Surelog as a standalone binary and use `read_uhdm` in the plugin.

YikeZhou, 2023-10-21:
Thanks! I'll try that.

alaindargelas, 2023-10-26:
@YikeZhou , this will be fixed when https://github.com/chipsalliance/Surelog/pull/3915 gets merged here.
You'll be able to invoke:
read_systemverilog -lowmem ......

Note 2 things about the -lowmem command, it indeed reduces the pic memory and you can use it in conjunction with -mp to use multiprocess compilation.

It has an advantage over the -defer -link commands in that it respects the order of packages compilation.

 You can also do as suggested above, run Surelog first with that option, then load the result .uhdm in Yosys with the read_uhdm command.

That, of course involves the use of a script outside of Yosys or a Yosys.ys that looks like:
```
plugin -i systemverilog.so

`<path to surelog>/surelog -lowmem ......    << Use the ` to invoke an executable from yosys

read_uhdm <path to surelog.uhdm>
```

alaindargelas, 2023-10-26:
Fixed by #2109

closed 2023-10-26 by alaindargelas