chipsalliance/synlig #2099: Latch inferred on function local variables

sifferman, 2023-10-24:
Hello. I am running into a bug where function local variables are inferred as latches.

Version: 2023-10-11-ef5c0bb

Example:

```systemverilog
module latch_test (
    input logic a,
    output logic [31:0] b
);

    function logic [31:0] func();
        logic [31:0] data = 4;
        return data;
    endfunction

    always_comb begin
        b = 0;
        if (a)
            b = func();
    end

endmodule;
```

kgugala, 2023-10-24:
Hi @sifferman, thanks for reporting

`always_comb` will infer a sensitivity list and will fire only if a signal in this sensitivity list change. I suppose that in your example the sensitivity list will be limited to `a` signal only so the `always_comb` logic will be fired only when the `a` signal changes. I suppose that in your case, the latch is inferred here:

```systemverilog
        if (a)
            b = func();
```
and just reported in a way it suggests it is inferred on a local variable. Can you try rewriting the code to:

```systemverilog
always_comb begin
        if (a)
            b = func();
        else
            b = 0;
    end
```
and checking it.

Also, at what stage you get the error (parsing SV, some optimization pass)?
Are you able to dump a netlist just before the error happens and share it here?

alaindargelas, 2023-10-26:
1) The error happens in the:
Executing PROC_DLATCH pass ....

2 ) The error is a safety net to make sure no latch got inferred in a always_comb block, which it catches nicely here.

3) Rewriting the code as suggested above does not fix the problem, the latch is created in any case.

This is the AST:

```
    AST_MODULE </home/alain/synlig/alain/tests/scratchpad/dut.sv:14.1-31.10> [0x555557d13e00] str='\top'
      AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:15.17-15.18> [0x555557c96d50] str='\a' input logic port=1 multirange=[ 0 1 ] multirange_swapped=[ 0 ]
        AST_RANGE <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557c35680] basic_prep range=[0:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557ce51e0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557e71fb0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
      AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:16.25-16.26> [0x555557a7fff0] str='\b' output logic port=2 multirange=[ 0 32 ] multirange_swapped=[ 0 ]
        AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:16.18-16.24> [0x555557cd5de0] basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557fcc0f0] bits='00000000000000000000000000011111'(32) signed basic_prep range=[31:0] int=31 in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557f73150] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
      AST_FUNCTION </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.5-22.16> [0x555557baa3d0] str='\func'
        AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.14-19.26> [0x555557d39920] str='\func' logic
          AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.20-19.26> [0x555557da8540] in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557c0b3a0] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b07810] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
        AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557c82990] str='\data' logic
          AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.15-20.21> [0x555557ef0110] in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x5555581f9390] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557e3f990] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
        AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.21> [0x555557bd2f00]
          AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.21> [0x555557fbca80] str='$unnamed_block$0'
            AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x5555581526f0] str='\data' logic
              AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.15-20.21> [0x55555764d1f0] in_param
                AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x55555777c790] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
                AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557770ef0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
            AST_ASSIGN_LE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.30> [0x555557f07d30]
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557d77ef0] str='\data' logic in_lvalue
              AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x55555765aad0] bits='00000000000000000000000000000100'(32) signed range=[31:0] int=4
            AST_ASSIGN_EQ </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.15> [0x555557638510]
              AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557ca3e50] str='\func' in_lvalue
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.16-21.20> [0x55555765b1e0] str='\data'
      AST_ALWAYS </home/alain/synlig/alain/tests/scratchpad/dut.sv:24.5-29.8> [0x555557b897e0]
        ATTR \always_comb:
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x55555809f000] bits='00000000000000000000000000000001'(32) range=[31:0] int=1 in_param
        AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:24.17-29.8> [0x555557cc2fb0]
          AST_ASSIGN_EQ </home/alain/synlig/alain/tests/scratchpad/dut.sv:25.9-25.14> [0x555557c69b50]
            AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:25.9-25.10> [0x555557f15850] str='\b' in_lvalue
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557c66b80] bits='00000000000000000000000000000000'(32) signed range=[31:0]
          AST_CASE </home/alain/synlig/alain/tests/scratchpad/dut.sv:26.9-27.24> [0x555557e99770]
            AST_REDUCE_BOOL <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557f5caa0]
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:26.13-26.14> [0x555557bb5590] str='\a'
            AST_COND <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557a733f0]
[... cut ...] -0.0> [0x555558103fd0] bits='1'(1) range=[0:0] int=1
              AST_BLOCK <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x5555580d9ef0]
                AST_ASSIGN_EQ </home/alain/synlig/alain/tests/scratchpad/dut.sv:27.13-27.23> [0x5555580416b0]
                  AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:27.13-27.14> [0x5555581fccc0] str='\b' in_lvalue
                  AST_FCALL </home/alain/synlig/alain/tests/scratchpad/dut.sv:27.17-27.23> [0x555557e08c20] str='\func'

```

alaindargelas, 2023-10-26:
The AST after simplification might shade some light
```
Dumping AST after simplification:
    AST_MODULE </home/alain/synlig/alain/tests/scratchpad/dut.sv:14.1-31.10> [0x555557caf380] str='\top' basic_prep
      AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:15.17-15.18> [0x555557b73290] str='\a' input logic basic_prep port=1 range=[0:0] multirange=[ 0 1 ] multirange_swapped=[ 0 ]
        AST_RANGE <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557aa8470] basic_prep range=[0:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557d6fac0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557a63570] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
      AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:16.25-16.26> [0x555557d6a010] str='\b' output logic reg basic_prep port=2 range=[31:0] multirange=[ 0 32 ] multirange_swapped=[ 0 ]
        AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:16.18-16.24> [0x555557a6f990] basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557daff70] bits='00000000000000000000000000011111'(32) signed basic_prep range=[31:0] int=31 in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557c98700] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
      AST_FUNCTION </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.5-22.16> [0x555557db4f40] str='\func'
        AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.14-19.26> [0x555557bab1f0] str='\func' logic
          AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:19.20-19.26> [0x555557ce3170] in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b5af10] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557ccf960] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
        AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557bebfa0] str='\data' logic
          AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.15-20.21> [0x555557c954f0] in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557cac140] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557befef0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
        AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.21> [0x555557d32ee0]
          AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.21> [0x555557d6e700] str='$unnamed_block$0'
            AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557d48820] str='\data' logic
              AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.15-20.21> [0x555557c37010] in_param
                AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557d17e80] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
                AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557d6a6e0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
            AST_ASSIGN_LE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.30> [0x555557db1c50]
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557c91f70] str='\data' logic in_lvalue
              AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557c8b510] bits='00000000000000000000000000000100'(32) signed range=[31:0] int=4
            AST_ASSIGN_EQ </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.9-21.15> [0x555557dad010]
              AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b58fc0] str='\func' in_lvalue
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:21.16-21.20> [0x555557c4c0f0] str='\data'
      AST_ALWAYS </home/alain/synlig/alain/tests/scratchpad/dut.sv:24.5-29.8> [0x555557a68c40] basic_prep
        ATTR \always_comb:
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b5be50] bits='00000000000000000000000000000001'(32) basic_prep range=[31:0] int=1 in_param
        AST_BLOCK </home/alain/synlig/alain/tests/scratchpad/dut.sv:24.17-29.8> [0x555557b6f510] basic_prep
          AST_ASSIGN_EQ </home/alain/synlig/alain/tests/scratchpad/dut.sv:25.9-25.14> [0x555557bcf9b0] basic_prep
            AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:25.9-25.10> [0x555557d6f8d0 -> 0x555557d6a010] str='\b' basic_prep in_lvalue
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b781f0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0]
          AST_CASE </home/alain/synlig/alain/tests/scratchpad/dut.sv:26.9-27.24> [0x555557a78910] basic_prep
            AST_REDUCE_BOOL <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557baa700] basic_prep
              AST_IDENTIFIER </home/alain/synlig/alain/tests/scratchpad/dut.sv:26.13-26.14> [0x555557b89450 -> 0x555557b73290] str='\a' basic_prep
            AST_COND <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557ca5650] basic_prep
[... cut ...]
      AST_WIRE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.22-20.26> [0x555557ab4200] str='\func$func$/home/alain/synlig/alain/tests/scratchpad/dut.sv:27$1.$unnamed_block$0.data' logic reg basic_prep range=[31:0]
        AST_RANGE </home/alain/synlig/alain/tests/scratchpad/dut.sv:20.15-20.21> [0x555557aad9a0] basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557ca91d0] bits='00000000000000000000000000011111'(32) signed basic_prep range=[31:0] int=31 in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557b388d0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
```

alaindargelas, 2023-10-26:
Comparatively, if simplifying the testcase using Verilog style functions and using the Verilog Yosys parser (read_verilog):

```
module top (
    input logic a,
    output int b
);

    function int myfunc;
        myfunc = 4;
    endfunction

    always_comb begin
        b = 0;
        if (a)
            b = myfunc();    
    end
endmodule
```

The simplified AST (that generates the correct circuit with no latches) is:

```
Dumping AST after simplification:
    AST_MODULE <alain/tests/scratchpad/dut.sv:14.1-30.10> [0x55555754ff10] str='\top' basic_prep
      AST_WIRE <alain/tests/scratchpad/dut.sv:15.17-15.18> [0x555557550080] str='\a' input logic basic_prep port=1 range=[0:0]
      AST_WIRE <alain/tests/scratchpad/dut.sv:16.16-16.17> [0x555557550230] str='\b' output logic reg signed basic_prep port=2 range=[31:0]
        AST_RANGE <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x5555575503b0] basic_prep range=[31:0] in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550530] bits='00000000000000000000000000011111'(32) signed basic_prep range=[31:0] int=31 in_param
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550720] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
      AST_FUNCTION <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550e10] str='\myfunc'
        AST_WIRE <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550ff0] str='\myfunc' reg signed
          AST_RANGE <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x5555575508f0] in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550a70] bits='00000000000000000000000000011111'(32) signed range=[31:0] int=31 in_param
            AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557550c40] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
        AST_ASSIGN_EQ <alain/tests/scratchpad/dut.sv:20.9-20.19> [0x5555575514f0]
          AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:20.9-20.15> [0x555557551170] str='\myfunc' in_lvalue
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:20.18-20.19> [0x555557551340] bits='00000000000000000000000000000100'(32) signed range=[31:0] int=4
      AST_ALWAYS <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557561e90] basic_prep
        ATTR \always_comb:
          AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557561fe0] bits='00000000000000000000000000000001'(32) basic_prep range=[31:0] int=1 in_param
        AST_BLOCK <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557562130] basic_prep
          AST_BLOCK <alain/tests/scratchpad/dut.sv:23.17-28.8> [0x5555575622e0] basic_prep
            AST_ASSIGN_EQ <alain/tests/scratchpad/dut.sv:24.9-24.14> [0x5555575627b0] basic_prep
              AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:24.9-24.10> [0x555557562460 -> 0x555557550230] str='\b' basic_prep in_lvalue
              AST_CONSTANT <alain/tests/scratchpad/dut.sv:24.13-24.14> [0x555557562600] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0]
            AST_CASE <alain/tests/scratchpad/dut.sv:25.9-26.26> [0x555557562a80] basic_prep
              AST_REDUCE_BOOL <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x5555575630b0] basic_prep
                AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:25.13-25.14> [0x555557562900 -> 0x555557550080] str='\a' basic_prep
              AST_COND <alain/tests/scratchpad/dut.sv:25.13-25.14> [0x555557562ef0] basic_prep
                AST_CONSTANT <alain/tests/scratchpad/dut.sv:0.0-0.0> [0x555557562d50] bits='1'(1) basic_prep range=[0:0] int=1
                AST_BLOCK <alain/tests/scratchpad/dut.sv:25.15-26.26> [0x555557562bd0] basic_prep
                  AST_ASSIGN_EQ <alain/tests/scratchpad/dut.sv:26.13-26.25> [0x5555575635a0] basic_prep
                    AST_IDENTIFIER <alain/tests/scratchpad/dut.sv:26.13-26.14> [0x5555575632a0 -> 0x555557550230] str='\b' basic_prep in_lvalue
                    AST_CONSTANT <alain/tests/scratchpad/dut.sv:26.17-26.23> [0x555557563420] bits='00000000000000000000000000000100'(32) signed basic_prep range=[31:0] int=4
```

The first thing that strikes me is that with the SystemVerilog AST, we have a **AST_ASSIGN_LE** operator.
With the Verilog AST, all assigns are AST_ASSIGN_EQ. That is a first problem to look at.

alaindargelas, 2023-11-10:
Fixed in https://github.com/alaindargelas/synlig/tree/stable_rs

closed 2023-11-10 by alaindargelas