chipsalliance/synlig #2100: Referencing enum in package

sifferman, 2023-10-24:
Hello. I am getting a bug where I cannot access enum values while inside a package. However, the error disappears if I add the package scope.

Version: 2023-10-11-ef5c0bb

Example:

```systemverilog
package pkg;

    typedef enum logic {
        TEST_1,
        TEST_2
    } test_t;

    function test_t swap(test_t test);
        unique case (test)
            TEST_1: return TEST_2; // works with pkg::TEST_1
            TEST_2: return TEST_1; // works with pkg::TEST_2
        endcase
    endfunction

endpackage

module enum_test (
    input pkg::test_t a,
    output pkg::test_t b
);
    always_comb b = pkg::swap(a);
endmodule
```

Error:

```plain
ERROR: Failed to resolve identifier \TEST_1 for width detection!
```

alaindargelas, 2023-10-27:
Needs #2114 to debug

alaindargelas, 2023-11-14:
Fixed by #2147

alaindargelas, 2023-11-14:
Merged in https://github.com/alaindargelas/synlig/tree/stable_rs

closed 2023-11-14 by alaindargelas