chipsalliance/synlig #2104: bash: - : invalid option while doing make install

gkamendje, 2023-10-25:
I cannot compile the latest version 2023-10-24-dd28e6d on Centos7. 
```
LSB Version:    :core-4.1-amd64:core-4.1-noarch:cxx-4.1-amd64:cxx-4.1-noarch:desktop-4.1-amd64:desktop-4.1-noarch:languages-4.1-amd64:languages-4.1-noarch:printing-4.1-amd64:printing-4.1-noarch
Distributor ID: CentOS
Description:    CentOS Linux release 7.9.2009 (Core)
Release:        7.9.2009
Codename:       Core
```

I get the following error messages which seem to be related to the line `override .SHELLFLAGS := -e -u -o pipefail -O nullglob -O extglob -O globstar -c` in the Makefile 

```
bash: - : invalid option
bash: - : invalid option
▖
bash: - : invalid option
▌ 
▌
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
bash: - : invalid option
▌ 
▘
bash: - : invalid option
bash: - : invalid option
bash: - : invalid option
bash: - : invalid option
bash: - : invalid option
bash: - : invalid option
bash: - : invalid option
Makefile:450: warning: overriding recipe for target `&'
Makefile:450: warning: ignoring old recipe for target `&'
bash: - : invalid option

```

gkamendje, 2023-10-25:
If I set the following in the Makefile `override .SHELLFLAGS := -v` and subsequently run `make build@systemverilog-plugin -n`
I get the following result. 
```
bash: - : invalid option
Usage:  bash [GNU long option] [option] ...
        bash [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --protected
        --rcfile
        --rpm-requires
        --restricted
        --verbose
        --version
Shell options:
        -irsD or -c command or -O shopt_option          (invocation only)
        -abefhkmnptuvxBCHP or -o option
```
I have tried setting the shell to `/bin/bash/` and /bin/sh` but does not work either.

gkamendje, 2023-10-25:
Using `override .SHELLFLAGS := -euo pipefail -O nullglob -O extglob -O globstar -c` and running `make build@systemverilog-plugin` I get the following 
```
bash: line 0: bash: for f in ; do [[ "$f" =~ ^[A-Z][A-Z0-9_]*$ ]] || echo "$f"; done; : invalid option name
bash: line 0: bash: printf '%s:\n' 'CFG_BUILD_TYPE' 'CFG_BUILD_DIR' 'CFG_OUT_DIR' 'CC' 'CXX' 'LD' 'CPPFLAGS' 'CXXFLAGS' 'LDFLAGS' 'LDLIBS' | wc -L: invalid option name
▖
bash: line 0: bash: printf '%-s\t%s' 'Date:' "$(date --iso-8601=seconds)": invalid option name
▌ 
▌
bash: line 0: bash: printf '%-s\t%s' 'CFG_BUILD_TYPE:' 'release' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CFG_BUILD_DIR:' 'build/release/' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CFG_OUT_DIR:' 'out/release/' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CC:' 'cc' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CXX:' 'c++' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'LD:' '' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CPPFLAGS:' '' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'CXXFLAGS:' '' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'LDFLAGS:' '' : invalid option name
▌ 
bash: line 0: bash: printf '%-s\t%s' 'LDLIBS:' '' : invalid option name
▌ 
▘
bash: line 0: bash: mkdir -p /home/gkamendje/tmp_compile/UHDM/synlig/build/: invalid option name
bash: line 0: bash: ln -fs -T release /home/gkamendje/tmp_compile/UHDM/synlig/build/current: invalid option name
bash: line 0: bash: mkdir -p /home/gkamendje/tmp_compile/UHDM/synlig/out/: invalid option name
bash: line 0: bash: ln -fs -T release /home/gkamendje/tmp_compile/UHDM/synlig/out/current: invalid option name
bash: line 0: bash: mkdir -p build/release/: invalid option name
bash: line 0: bash: find /home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/ -path '*/.*' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/build' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/tests' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/third_party/tests' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/third_party/UHDM/tests' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/surelog/third_party/antlr4/runtime/Cpp/demo' -prune -o -name 'compile_commands.json' -o -type f -print | sed "s/ /\\\\ /g" : invalid option name
bash: line 0: bash: find /home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/ -path '*/.*' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/abc' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/docs' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/tests' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/manual' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/examples' -prune -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/yosys' -o -path '/home/gkamendje/tmp_compile/UHDM/synlig/third_party/yosys/*/yosys-*' -o -name '*.o' -o -name '*.d' -o -type f -print | sed "s/ /\\\\ /g" : invalid option name
Makefile:451: warning: overriding recipe for target `&'
Makefile:451: warning: ignoring old recipe for target `&'
bash: line 0: bash: mkdir -p /home/gkamendje/tmp_compile/UHDM/synlig/build/release/surelog/: invalid option name
make: *** [/home/gkamendje/tmp_compile/UHDM/synlig/build/release/surelog/] Error 2

```
It seems that for some reason some variables are not expanded.

gkamendje, 2023-10-25:
CentOS7 uses bash 4.1. by default. I have compiled Bash 5.2 and modified the shell variable in the Makefile to point to the new version of bash but the problem persists 
```
$ make list
/home/gkamendje/tmp_compile/bash-5.2/bash: - : invalid option
/home/gkamendje/tmp_compile/bash-5.2/bash: - : invalid option
/home/gkamendje/tmp_compile/bash-5.2/bash: - : invalid option
/home/gkamendje/tmp_compile/bash-5.2/bash: - : invalid option
```

alaindargelas, 2023-10-25:
@gkamendje, I faced the exact same problem, that is why I create a CMake that works on Ubuntu, CentOS (Validated) and eventually MacOS and Windows.
It is pending merge:
https://github.com/chipsalliance/synlig/pull/2089

alaindargelas, 2023-10-25:
Please try the pending PR and see if it would solve your issues.

gkamendje, 2023-10-25:
Running with 
```
commit 2fbf4a982327cfaab2e62070e8428f8cbe385a55 (HEAD -> alternative_cmake_build, origin/alternative_cmake_build)
Author: Alain Dargelas <alain.dargelas@gmail.com>
Date:   Tue Oct 24 10:20:15 2023 -0700

    symplify flags to mininum
```
I still see the same problem. 

There used to be a shell script `build_binaries.sh` in the previous versions that worked for me pretty well. That script was unfortunately removed

alaindargelas, 2023-10-25:
Did you do
make -f cmake-makefike 

After cleaning all previous build and compiled objects?
You need to start clean to use the alternative build system

gkamendje, 2023-10-25:
Oh thanks for the hint. The errors are gone. Now I get this. 
```
$ make -f cmake-makefile 
cmake -DCMAKE_BUILD_TYPE=Release -DCPU_CORES=8 -DYOSYS_CONFIG= -DYOSYS_PATH= -DCMAKE_INSTALL_PREFIX=/usr/local -DCMAKE_RULE_MESSAGES=on -S . -B build
CMake Error: The source directory "/home/gkamendje/tmp_compile/UHDM/synlig_dargelas/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [run-cmake-release] Error 1
```

gkamendje, 2023-10-25:
Are these two  commands still required? 
 ```
git submodule sync
git submodule update --init --recursive third_party/{surelog,yosys}
```

alaindargelas, 2023-10-25:
Yes they are. Same checkout process, just different build

gkamendje, 2023-10-25:
Thanks a lot. Now I am able to compile using cmake 3.2.7.

pgielda, 2023-10-25:
Some official comments:

1. The only *officially supported* and tested Linux platform is currently debian:trixie.
2. Even the oldest Debian that you can still get packages to (debian:buster) has a newer bash than centos7 and that part of the buildsystem works fine there (but cmake seems to be too old)
3. I've just checked that Synlig builds as is on new Gentoo and Arch. Having said that this is not officially supported, I just checked for the sake of checking. I am pretty sure it also builds on e.g. Ubuntu and Fedora but have no time to check.
4. I've just checked on Gentoo with forced old bash 4.2 and the buildsystem works fine, so in my opinion its not the version of bash, its something different about CentOS that makes it incompatible. Having said that I am not really using CentOS so hard to say.
5. I can reproduce your problem on a container with centos7. So far this is the only container that I was able to reproduce this problem with...
6. I was able to get it further on centos7 by commenting out "override .SHELLFLAGS" line in Makefile, but it got stuck somewhere else, but if you want to you can try
7. If you care about that platform can provide a (minimal) patch that fixes centos7 and does not break debian:trixie we will definitely try to review and merge that patch.
8. Switching the buildsystem to something else just because it does not work on centos7 is not considered a minimal patch, currently not really something on the roadmap to be officially included and supported in Synlig but feel free to use @alaindargelas branches if they work for you.
This does not mean that it will never be switched or that there never will be an alternative official way to do things, just not now, as we have more pressing things to work on.

As a side note Debian can also be installed on Windows ("WSL") as well as run as a container (e.g. I do not personally use Debian, I just run it in a container when I need it.)

pgielda, 2023-10-25:
Oh, I just noticed you wrote bash-4.1 not 4.2, so I've re-tested with gentoo+bash-4.0 and it *still works*, even on bash older than 4.1...

hzeller, 2023-10-25:
Maybe the issue is somewhere else ?
 
The [lines that print stuff](https://github.com/chipsalliance/synlig/blob/main/Makefile#L223-L231) where apparently the error happens contains `▌`. Maybe replacing that with an ascii character will help ?

pgielda, 2023-10-25:
Actually I asked to remove those characters because vi in the CentOS container was not handling those characters properly and it was hard to edit files... I doubt this is the reason but who knows...

gkamendje, 2023-10-26:
Just for information, the unicode characters are not the source of the issue. This seems to be something buried deep into CentOs 7. I was able to compile with CentOs Stream 9 without problem (yes I know this Distro is not going anywhere).

pgielda, 2023-10-27:
Yes, I had a similar issue in a centos7 container (it was consuming 100% cpu and not doing anything after printing configuration when .SHELLFLAGS was commented out) but I was hoping its just my computer ;)
I was not expecting removing unicode characters will help to be honest.
Seems that @alaindargelas branch was reviewed and merged faster than I expected so I guess you can use the cmake approach for now.

alaindargelas, 2023-11-14:
Fixed by make -f cmake-makefile

closed 2023-11-14 by alaindargelas