chipsalliance/synlig #2126: Array literals have no effect within continuous assignments

YikeZhou, 2023-11-03:
## Steps to Reproduce

SystemVerilog file `top.sv`:
```systemverilog
module top(output logic[7:0] o);
   assign o = '{0:1, 3:1, 7:1, default:0};
endmodule
```

I used the Verilog backend of Yosys (`write_verilog`) to obtain a Verilog file for further use. Since I found
> Verilog output from Yosys is imported into Verilator and simulated.

in the [README of repo UHDM-integration](https://github.com/chipsalliance/UHDM-integration-tests#full-flow), I thought this was a sensible way to utilize this plugin.

Therefore, I ran:
```
yosys -p "plugin -i systemverilog; read_systemverilog top.sv; write_verilog"
```
Got:
```
 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.34+55 (git sha1 672375ed0, clang 14.0.6 -fPIC -Os)

-- Running command `plugin -i systemverilog; read_systemverilog top.sv; write_verilog' --

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file "/tmp/slpp_all/surelog.log".
[INF:CP0300] Compilation...
[INF:CP0303] /tmp/top.sv:1:1: Compile module "work@top".
[INF:CP0302] Compile class "work@mailbox".
[INF:CP0302] Compile class "work@process".
[INF:CP0302] Compile class "work@semaphore".
[INF:EL0526] Design Elaboration...
[NTE:EL0503] /tmp/top.sv:1:1: Top level module "work@top".
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 1.
[NTE:EL0510] Nb instances: 1.
[NTE:EL0511] Nb leaf instances: 1.
[INF:UH0706] Creating UHDM Model...
[INF:UH0707] Elaborating UHDM...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 5
Generating RTLIL representation for module `\top'.

2. Executing Verilog backend.

2.1. Executing BMUXMAP pass.

2.2. Executing DEMUXMAP pass.
/* Generated by Yosys 0.34+55 (git sha1 672375ed0, clang 14.0.6 -fPIC -Os) */
Dumping module `\top'.

(* cells_not_processed =  1  *)
(* src = "/tmp/top.sv:1.1-3.10" *)
module top(o);
  (* src = "/tmp/top.sv:1.30-1.31" *)
  output [7:0] o;
  wire [7:0] o;
  assign o = 8'h00;
endmodule

End of script. Logfile hash: c46a626fd2, CPU: user 0.07s system 0.03s, MEM: 33.88 MB peak
Yosys 0.34+55 (git sha1 672375ed0, clang 14.0.6 -fPIC -Os)
Time spent: 84% 2x read_systemverilog (0 sec), 14% 1x plugin (0 sec), ...
```
in which I found a `(* cells_not_processed =  1  *)` message and the value assigned to signal `o` became `8'h00`.

## Versions

- Yosys: Yosys 0.34+55 (git sha1 672375ed0, clang 14.0.6 -fPIC -Os)
- Synlig: [2023-11-01-b746752](https://github.com/chipsalliance/synlig/releases/tag/2023-11-01-b746752)

alaindargelas, 2023-11-03:
Try my feature branch:
https://github.com/alaindargelas/synlig/tree/stable_rs

It contains the fix:
https://github.com/chipsalliance/Surelog/pull/3922

Synlig main will get the fix at the next update of Surelog.

alaindargelas, 2023-11-10:
Fixed in https://github.com/alaindargelas/synlig/tree/stable_rs

closed 2023-11-10 by alaindargelas