chipsalliance/synlig #2134: Assignment to multidimensional vectors is broken

sifferman, 2023-11-08:
Hello.

I found that this

https://github.com/chipsalliance/synlig/tree/2023-10-29-9d4cf25

PR handles assignment to multidimensional vectors incorrectly: https://github.com/chipsalliance/synlig/pull/2072

Compare commits: https://github.com/chipsalliance/synlig/compare/2023-10-27-01111c5...2023-10-29-9d4cf25

```tcl
# vector_test.tcl

rename eval tcl_eval

yosys -import
plugin -i systemverilog
yosys -import

read_systemverilog vector_test.sv
prep

eval -set a 1 -set b 2 -set-undef vector_test
```

```systemverilog
// vector_test.sv

module vector_test(
    input logic [3:0] a,
    input logic [3:0] b,
    output logic [1:0][3:0] out
);

    typedef logic [3:0] logic4;

    logic4 [1:0] vector2x4;

    assign out = vector2x4;

    assign vector2x4[0] = a;
    assign vector2x4[1] = b;

endmodule
```

alaindargelas, 2023-11-08:
@sifferman to help diagnose, can you locally undo the section of the PR from line 4428 to 4451?

https://github.com/chipsalliance/synlig/pull/2072/files#diff-514c4dbb1f1744bd9f6c431e69ecff166762ef801400abec69d89c7b4dc80431L4438

This is the resulting code:
```
void UhdmAst::process_logic_var()
{
    current_node = make_ast_node(AST::AST_WIRE);
    current_node->is_logic = true;
    std::vector<AST::AstNode *> packed_ranges;   // comes before wire name
    std::vector<AST::AstNode *> unpacked_ranges; // comes after wire name
    // TODO: add const attribute, but it seems it is little more
    // then just setting boolean value
    // current_node->is_const = vpi_get(vpiConstantVariable, obj_h);
    visit_one_to_one({vpiTypespec}, obj_h, [&](AST::AstNode *node) {
        if (node->str.empty()) {
            // anonymous typespec, move the children to variable
            current_node->type = node->type;
            current_node->children = std::move(node->children);
            //copy_packed_unpacked_attribute(node, current_node);
        } else {
            auto wiretype_node = new AST::AstNode(AST::AST_WIRETYPE);
            wiretype_node->str = node->str;
            current_node->children.push_back(wiretype_node);
            current_node->is_custom_type = true;
        }
        current_node->is_signed = node->is_signed;
        delete node;
    });
    if (auto ref_typespec_h = vpi_handle(vpiTypespec, obj_h)) {
        if (auto typespec_h = vpi_handle(vpiActual, ref_typespec_h)) {
            visit_one_to_many({vpiRange}, typespec_h, [&](AST::AstNode *node) { packed_ranges.push_back(node); });
            vpi_release_handle(typespec_h);
        } else {
            visit_one_to_many({vpiRange}, obj_h, [&](AST::AstNode *node) { packed_ranges.push_back(node); });
        }
        vpi_release_handle(ref_typespec_h);
    } else {
        visit_one_to_many({vpiRange}, obj_h, [&](AST::AstNode *node) { packed_ranges.push_back(node); });
    }
    visit_default_expr(obj_h);
    add_multirange_wire(current_node, packed_ranges, unpacked_ranges);
}
```

That change restores the multi-dimension of  vector2x4.
What else is incorrect?

alaindargelas, 2023-11-14:
@sifferman , can you answer the question above?

sifferman, 2023-11-14:
Thanks for following up, @alaindargelas.

I cloned [alaindargelas/synlig/stable_rs](https://github.com/alaindargelas/synlig/tree/stable_rs), undid the specified section from the PR, then built.

Now the example I provided works, but there are still issues. Here are a few test cases that still do not work:

## Vector Inside Struct

```systemverilog
module vector_test(
    input logic [3:0] a,
    input logic [3:0] b,
    output logic [1:0][3:0] out
);

    struct packed {
        logic [1:0][3:0] vector2x4;
    } s;

    assign s.vector2x4[0] = a;
    assign s.vector2x4[1] = b;

    assign out = s.vector2x4;

endmodule
```
```plain
AST_MODULE <vector_test.sv:2.1-17.10> [0x561c02a60110] str='\vector_test'
  AST_WIRE <vector_test.sv:3.23-3.24> [0x561c02759380] str='\a' input logic port=1 multirange=[ 0 4 ] multirange_swapped=[ 0 ]
    AST_RANGE <vector_test.sv:3.17-3.22> [0x561c0297bec0] basic_prep range=[3:0] in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02849800] bits='00000000000000000000000000000011'(32) signed basic_prep range=[31:0] int=3 in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c0293caa0] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
  AST_WIRE <vector_test.sv:4.23-4.24> [0x561c0276a510] str='\b' input logic port=2 multirange=[ 0 4 ] multirange_swapped=[ 0 ]
    AST_RANGE <vector_test.sv:4.17-4.22> [0x561c028c9a10] basic_prep range=[3:0] in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02336920] bits='00000000000000000000000000000011'(32) signed basic_prep range=[31:0] int=3 in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02333e00] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0] in_param
  AST_WIRE <vector_test.sv:5.29-5.32> [0x561c0277aaf0] str='\out' output logic port=3 multirange=[ 0 2 0 4 ] multirange_swapped=[ 0 0 ]
    AST_RANGE <vector_test.sv:0.0-0.0> [0x561c028f7fa0] range=[7:0] in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c027600d0] bits='00000000000000000000000000000111'(32) signed range=[31:0] int=7 in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02aa0930] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
  AST_STRUCT <vector_test.sv:10.7-10.8> [0x561c02822c70] str='\s' basic_prep
    AST_STRUCT_ITEM <vector_test.sv:9.26-9.35> [0x561c0299fd50] str='vector2x4' logic basic_prep range=[7:0] multirange=[ 0 2 0 4 ] multirange_swapped=[ 0 0 ]
  AST_ASSIGN <vector_test.sv:12.12-12.30> [0x561c028c0220]
    AST_IDENTIFIER <vector_test.sv:12.14-12.23> [0x561c029b8840] str='\s' basic_prep in_lvalue
      AST_RANGE <vector_test.sv:0.0-0.0> [0x561c029aa610] in_lvalue
        AST_ADD <vector_test.sv:0.0-0.0> [0x561c0286da90] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c029272e0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02a5c750] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
        AST_ADD <vector_test.sv:0.0-0.0> [0x561c0285cf70] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02a3a6e0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02a59040] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
    AST_IDENTIFIER <vector_test.sv:12.29-12.30> [0x561c02844080] str='\a'
  AST_ASSIGN <vector_test.sv:13.12-13.30> [0x561c02769200]
    AST_IDENTIFIER <vector_test.sv:13.14-13.23> [0x561c02aa2610] str='\s' basic_prep in_lvalue
      AST_RANGE <vector_test.sv:0.0-0.0> [0x561c0296d1c0] in_lvalue
        AST_ADD <vector_test.sv:0.0-0.0> [0x561c0281ea00] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02931590] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02a9d120] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1 in_lvalue
        AST_ADD <vector_test.sv:0.0-0.0> [0x561c02982920] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c0234b7c0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_lvalue
          AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02909b30] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1 in_lvalue
    AST_IDENTIFIER <vector_test.sv:13.29-13.30> [0x561c02798b90] str='\b'
  AST_ASSIGN <vector_test.sv:15.12-15.29> [0x561c02853930]
    AST_IDENTIFIER <vector_test.sv:15.12-15.15> [0x561c0279d640] str='\out' in_lvalue
    AST_IDENTIFIER <vector_test.sv:15.18-15.19> [0x561c02a5ef40] str='\s' basic_prep
      AST_RANGE <vector_test.sv:0.0-0.0> [0x561c027df530]
        AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02938bd0] bits='00000000000000000000000000000111'(32) signed range=[31:0] int=7
        AST_CONSTANT <vector_test.sv:0.0-0.0> [0x561c02886ff0] bits='00000000000000000000000000000000'(32) signed range=[31:0]
  AST_WIRE <vector_test.sv:0.0-0.0> [0x561c029d3a40] str='\s' logic range=[-1:0] multirange=[ 0 8 ] multirange_swapped=[ 0 ]
    ATTR \wiretype:
[... 64 lines cut ...]
    AST_IDENTIFIER <vector_test.sv:14.14-14.23> [0x55bc42df1720] str='\s' basic_prep in_lvalue
      AST_RANGE <vector_test.sv:0.0-0.0> [0x55bc42c83ef0] in_lvalue
        AST_ADD <vector_test.sv:0.0-0.0> [0x55bc4283cf40] in_lvalue
[... cut ...]
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x55bc42e191b0] bits='00000000000000000000000000000001'(32) signed range=[31:0] int=1 in_param
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x55bc42f443e0] bits='00000000000000000000000000000000'(32) signed range=[31:0] in_param
    AST_RANGE <vector_test.sv:0.0-0.0> [0x55bc42de1b40] basic_prep range=[1:0]
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x55bc42d57e90] bits='00000000000000000000000000000001'(32) signed basic_prep range=[31:0] int=1
      AST_CONSTANT <vector_test.sv:0.0-0.0> [0x55bc42e34600] bits='00000000000000000000000000000000'(32) signed basic_prep range=[31:0]
```

alaindargelas, 2023-11-18:
Making slow progress on this case,
I fixed the bugs in the plugin code that fix the size of the vector. The hierarchical reference of the struct member still eludes me, the code seems to not support struct member sizing while walking the hierarchical path

alaindargelas, 2023-11-28:
Fixed in https://github.com/alaindargelas/synlig/tree/stable_rs

closed 2023-11-28 by alaindargelas