chipsalliance/synlig #2228: Parsing open-source processor core RIDECORE crashes.

CrazybinaryLi, 2023-12-22:
The URL of RIDECORE is: https://github.com/ridecore/ridecore
Read RIDECORE RTL by yosys script. 
Using command: 
plugin -i systemverilog; 
read_systemverilog -formal ... ;
prep -top top ;
...

The process crashes !!!

alaindargelas, 2023-12-22:
Please add here the complete yosys .ys script file.

alaindargelas, 2023-12-26:
The file FV/property.sv does not exist in the repo. Please attach it.
Also, the project file does not correspond to the actual repo, it looks like the project you attached is for a hard copy of the project with modified directory structure....

CrazybinaryLi, 2023-12-26:
Yes, I pulled the RIDECORE project into my local directory and wrote another file locally that contains SystemVerilog assertions. You can try downloading RIDECORE and writing an SVA to check if synlig can parse it.

alain-rs, 2023-12-26:
Please provide a file, even a simple one, else no one will take a look at this issue.
Generally you have to provide simple and accurate instructions to reproduce an issue, else it won't be worked on.

CrazybinaryLi, 2023-12-26:
I have updated the script file.

alaindargelas, 2023-12-27:
Fixed in https://github.com/chipsalliance/synlig/pull/2236

closed 2023-12-27 by alaindargelas

CrazybinaryLi, 2023-12-30:
This issue has not been fully resolved! The crash is caused by the excessive use of physical memory when using the "plugin -i systemverilog" in the .ys script. Insufficient memory leads to the crash. Currently, this problem still persists.

alain-rs, 2023-12-30:
You probably didn't check out the branch correctly.
Anyhow since I closed the issue,  the main branch has been updated.
Checkout a clean clone from scratch that should fix your local issue