chipsalliance/synlig #2310: version `GLIBC_2.35' not found

rodrigomelo9, 2024-02-04:
I'm trying on Ubuntu 20.04, but I also tried `ubuntu:latest` (docker image).

I installed Yosys from sources:
```
sudo apt install build-essential ca-certificates clang bison flex libreadline-dev gawk tcl-dev libffi-dev git graphviz xdot pkg-config python3 libboost-system-dev libboost-python-dev libboost-filesystem-dev zlib1g-dev
git clone --depth 1 https://github.com/YosysHQ/yosys && cd yosys && make
sudo make install
```
And synlig as following:
```
sudo apt install curl jq tar wget
mkdir synlig && cd synlig && curl https://api.github.com/repos/chipsalliance/synlig/releases/latest | jq .assets[1] | grep "browser_download_url" | grep -Eo 'https://[^\"]*' | xargs wget -O - | tar -xz 
sudo ./install_plugin.sh
```

When I run:
```
$ yosys -Q -p 'plugin -i systemverilog'

-- Running command `plugin -i systemverilog' --
ERROR: Can't load module `./systemverilog': /usr/local/bin/../share/yosys/plugins/systemverilog.so: cannot open shared object file: Permission denied
```

And with sudo:
```
$ sudo yosys -Q -p 'plugin -i systemverilog'

-- Running command `plugin -i systemverilog' --
ERROR: Can't load module `./systemverilog': /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.35' not found (required by /usr/local/bin/../share/yosys/plugins/systemverilog.so)
```

rodrigomelo9, 2024-02-04:
I also tried with `debian:stable-slim`:

```
sudo docker XXX yosys -Q -p "plugin -i systemverilog"
ERROR: Can't load module `./systemverilog': /usr/local/bin/../share/yosys/plugins/systemverilog.so: undefined symbol: _ZN5Yosys5RTLIL8IdString12global_cacheE
```

alaindargelas, 2024-02-05:
Check out a fresh clone, 
Follow the way the GitHub action checks out the repo,
Then make -f cmake-makefile

Do not use your yosys build, use the yosys build that is checked out by the Synlig repo.

rodrigomelo9, 2024-02-05:
What you propose is not what the Installation section says. Is it a workaround or instructions are outdated? To be honest, for me, the best approach would be to install all from sources, instead of depending on pre-built packages. I would like to use Yosys mainstream instead of a, probably outdated, version.

pgielda, 2024-02-05:
Plugins in Yosys are just shared libraries that are loaded in runtime. They do have to be compatible with the yosys version you have, and there is no clear split between the main binary and the plugin (aka an API) so things break.

Therefore:

The "installation guide" is for Debian Trixie and expects that you install a debian package (https://github.com/chipsalliance/synlig#installation)

The "from sources" installation guide is expecting you to use *our* sources from submodules (https://github.com/chipsalliance/synlig#installation-from-source).

Both are tested in a CI (but I can imagine a slim chance of sudden Debian breaking due to some change in a package).
We only try to maintain compatibility with the Debian package (because we can test against that and try to accommodate). We cannot be compatible with any commit you happen to clone -- its simply impossible to maintain. You can probably tweak your sources to get it to work, but you're on your own here...

pgielda, 2024-02-05:
There is no hidden agenda here, Synlig requires a pretty recent Yosys version, and debian:trixie is the first Debian to have a packaged version that is new enough.

We also provide fully binary releases, there is Yosys in those as well - those might work for you but no promises.

alaindargelas, 2024-02-05:
@rodrigomelo9 , if you prefer to compile from source:

https://github.com/chipsalliance/synlig?tab=readme-ov-file#installation-from-source

rodrigomelo9, 2024-02-07:
Ok, I will try to generate the plugin from sources.
Let me stress that, I think the problem is, the GLIBC version needed for the distributed (https://api.github.com/repos/chipsalliance/synlig/releases/latest) pre-built plugin (it is not compatible with nowadays stable versions, such as Ubuntu 20.04 or the current stable Debian).

pgielda, 2024-02-07:
What we could potentially do is to add an additional test that checks if the full binary release works on debian:bookworm, this might reveal this issue and then we can try to find a way to make it work... FYI @kgugala @tgorochowik

rodrigomelo9, 2024-02-09:
I successfully compiled from sources, what solved my problem. Additionally, I found that this issue (GLIB_2_35 not found) was involved in previous issues (#1776 #1832 ).

suzizecat, 2024-06-21:
Hello there, 

Chiming in to inform you that the kind of issues discussed here also happen on the yosys packaged in the [OSS Cad Suite](https://github.com/YosysHQ/oss-cad-suite-build/) which provides the latest stable Yosys (0.42 at the time of writing) .

The README mention 

> The required Yosys version is 0.33 or later.

Which is fairly different from "The Required Yosys version is the one bundled" 

With Yosys 0.42, the plugin loading fails with 
```console
yosys> plugin -i systemverilog
ERROR: Can't load module `./systemverilog': /home/xxx/opt/oss-cad-suite/lib/../share/yosys/plugins/systemverilog.so: undefined symbol: _ZN5Yosys5RTLIL8IdString12global_cacheE
```

Now, if I understand correctly, the plugin system is "just" providing access points to Yosys internals that may (and have) change without particular notice which feels like an absolute pain to support on the plugin side.

If you want me to, I can open a new issue tracking this particular problem 🙂