chipsalliance/synlig #2322: Module not found

rodrigomelo9, 2024-02-09:
Using `yosys` and `synlig` compiled from sources, I'm trying to convert SV into V (without synthesis), as follows:
```
yosys -Q -p "
plugin -i systemverilog
read_systemverilog counter.sv
hierarchy -top counter
write_verilog -noattr counter.v
"
```
And it fails:
```
ERROR: Module `counter' not found!

-- Running command `
plugin -i systemverilog
read_systemverilog counter.sv
hierarchy -top counter
write_verilog -noattr counter.v
' --
    -noassert
        ignore assert() statements
    -debug
        alias for -dump_ast1 -dump_ast2 -dump_vlog1 -dump_vlog2
```
Some comments:
* Not sure if I'm doing things 100% right, but it worked 5 months ago (and yes, `counter` exists). Moreover, I'm successfully doing the same with VHDL (using `ghdl-yosys-plugin`).
* I know in this case specifying the top level is not needed, but I'm looking for a general-purpose solution where multiple files will be specified.
* Why, after `Running command`, a list of options is displayed? I think it was not there 5 months ago and it doesn't happen with `ghdl-yosys-plugin`.

alaindargelas, 2024-02-09:
Make sure you use the yosys built by Synlig, not any other Yosys you built yourself. The Yosys built by Synlig has the plugin installed. It looks like your plugin and Yosys are not compatible.
Synlig has 100s of regressions passing on GitHub actions, none show the behavior you are seeing

rodrigomelo9, 2024-02-10:
I see... So, following "Installation" I can use the Yosys built by Debian, but with "Installation from source" I need a particular Yosys version... I will try.