chipsalliance/synlig #2376: Compatibility with Yosys 0.39

hzeller, 2024-03-21:
`struct Yosys::AST::AstNode` changed recent.y, so Synlig unfortunately will currently not compile with Yosys 0.39

 * `Yosys::AST::AstNode` does not have a `multirange_dimensions` anymore (or it is renamed)
 * Similarly `multirange_swapped`
 * Also `make_struct_member_range`

donn, 2024-07-22:
Are there any updates on this? Synlig is the only thing keeping us on 0.38…

kamilrakoczy, 2024-07-23:
Thank you for taking interest and working on it in: https://github.com/chipsalliance/synlig/pull/2496. While it fixes compilation with newer yosys, we also need to make all tests pass there.

Please note, we are also working on updating yosys in: https://github.com/chipsalliance/synlig/pull/2487 and in upcoming PRs. We took different approach there: we are updating yosys to next version that is breaking our tests and fixing issues one by one, so it might take a while while we will support upstream yosys.

Having said that, if you will manage to get all tests passed in your PR, I'm happy to merge it too.

donn, 2024-07-24:
I expect your approach to probably work better lol. I'll see what I can do on my end.

kamilrakoczy, 2024-08-07:
We managed to update compatibility to yosys 0.44 (current upstream): https://github.com/chipsalliance/synlig/pull/2510

Closing this issue at this point. If you will encourage any new problems with updated yosys version, feel free to open new issue.

closed 2024-08-07 by kamilrakoczy