chipsalliance/synlig #2377: ERROR: Assert `size >= 1' failed in /root/synlig/synlig/frontends/systemverilog/uhdm_ast.cc:751.

yosyssyn, 2024-03-21:
[image]
I am using the following commands to load the files:

Yosys 0.33 version
plugin -i systemverilog
read_systemverilog -defer file_name1.sv 
.read_systemverilog -defer file_name2.sv 
read_systemverilog -defer file_name3.sv 
read_systemverilog -defer top.sv

read_systemverilog -link
when I am linking with above command, I am getting the ERROR: Assert `size >= 1' failed in /root/synlig/synlig/frontends/systemverilog/uhdm_ast.cc:751.

alaindargelas, 2024-03-26:
All the warnings above the error are of great concern.
Please verify your design setup (Parameters).

yosyssyn, 2024-03-26:
hi
Thank you alaindargelas, I am trying to reevaluate the parameters from top file but the following error is coming 
[image]

alaindargelas, 2024-03-26:
Use read_systemverilog -P<name>=<value> ....rest of the command line

alaindargelas, 2026-05-13:
@yosyssyn, try https://github.com/alainmarcel/uhdm2rtlil