chipsalliance/synlig #2395: Windows support?

RogerSanders, 2024-04-01:
Yosys supports Windows, and Surelog supports Windows, but from what I can see, it appears synlig currently only provides support for compiling and deploying on Linux. Are there any plans to add Windows support? Are there any technical issues that would make supporting Windows problematic?

tgorochowik, 2024-04-03:
That's correct, Yosys does support windows, including plugins. The same goes for Surelog.

There are no targets for Windows in Synlig as we're not using Windows so it wasn't a high priority for us. That being said, we're not opposed to adding this if there's interest.

Would you be willing to give it a go and contribute a PR?

RogerSanders, 2024-04-06:
I'll probably make an attempt. I'm new to Yosys, but I like the idea of being able to leverage SystemVerilog natively under Windows. Once I've got some experience with the toolchain I'll have a shot at doing the port.

alaindargelas, 2024-04-06:
Your best bet is the CMakeLists.txt path.
You can augment the Synlig's one with what was done in Surelog's for Windows support.