chipsalliance/synlig #2430: Can't find top module when using plugin

zaun, 2024-04-27:
I've installed `yosys`. I've installed the `systemverilog` plugin from `https://github.com/chipsalliance/synlig`.

I can load the plugin, but then it can't find the `top` module.

```
yosys -p "plugin -i systemverilog" -p "read_systemverilog src/main.sv; synth_gowin -top top -json synthesis.json"

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.40 (git sha1 a1bb0255d65, clang++ 15.0.0 -fPIC -Os)

-- Running command `plugin -i systemverilog' --

-- Running command `read_systemverilog src/main.sv; synth_gowin -top top -json synthesis.json' --
    -noassert
        ignore assert() statements
    -debug
        alias for -dump_ast1 -dump_ast2 -dump_vlog1 -dump_vlog2

    -dump_ast1
        dump abstract syntax tree (before simplification)

    -dump_ast2
        dump abstract syntax tree (after simplification)

    -no_dump_ptr
        do not include hex memory addresses in dump (easier to diff dumps)

    -dump_vlog1
        dump ast as Verilog code (before simplification)

    -dump_vlog2
        dump ast as Verilog code (after simplification)

    -dump_rtlil
        dump generated RTLIL netlist

    -defer
        only read the abstract syntax tree and defer actual compilation
        to a later 'hierarchy' command. Useful in cases where the default
        parameters of modules yield invalid or not synthesizable code.
        Needs to be followed by read_systemverilog -link after reading
        all files.

    -link
        performs linking and elaboration of the files read with -defer

    -parse-only
        this parameter only applies to read_systemverilog command,
        it runs only Surelog to parse design, but doesn't load generated
        tree into Yosys.

    -formal
        enable support for SystemVerilog assertions and some Yosys extensions
        replace the implicit -D SYNTHESIS with -D FORMAL

1. Executing SYNTH_GOWIN pass.

1.1. Executing Verilog-2005 frontend: /Users/justinzaun/local/brew/bin/../share/yosys/gowin/cells_sim.v
Parsing Verilog input from `/Users/justinzaun/local/brew/bin/../share/yosys/gowin/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT1'.
Generating RTLIL representation for module `\LUT2'.
Generating RTLIL representation for module `\LUT3'.
Generating RTLIL representation for module `\LUT4'.

*** REMOVED A BUNCH OF LOGS ***

Generating RTLIL representation for module `\DCCG'.
Generating RTLIL representation for module `\FLASH96KA'.
Generating RTLIL representation for module `\MIPI_DPHY_RX'.
Generating RTLIL representation for module `\CLKDIVG'.
Successfully finished Verilog frontend.

1.3. Executing HIERARCHY pass (managing design hierarchy).
ERROR: Module `top' not found!
```

all that is in `src/main.sv` is a simple top module to count on some leds:

```
module top
(
  input CLK,
  output [5:0] LED
);

reg [23:0] wait_counter = 'd0;
reg [5:0] ledCounter = 0;

assign LED = ~ledCounter;

always @(posedge CLK) begin
  if (wait_counter < 13500000) begin
    wait_counter <= wait_counter + 'd1;
  end
  else begin
    wait_counter <= 'd0;
    ledCounter <= ledCounter + 'd1;
  end
end

endmodule
```

This same code WILL build with just Verilog, but I'm trying to get it to build as SystemVerilog as I want to start using SystemVerilog features.

```
yosys -p "read_verilog src/main.sv; synth_gowin -top top -json synthesis.json"   

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.40 (git sha1 a1bb0255d65, clang++ 15.0.0 -fPIC -Os)

-- Running command `read_verilog src/main.sv; synth_gowin -top top -json synthesis.json' --

1. Executing Verilog-2005 frontend: src/main.sv
Parsing Verilog input from `src/main.sv' to AST representation.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

2. Executing SYNTH_GOWIN pass.

2.1. Executing Verilog-2005 frontend: /Users/justinzaun/local/brew/bin/../share/yosys/gowin/cells_sim.v
Parsing Verilog input from `/Users/justinzaun/local/brew/bin/../share/yosys/gowin/cells_sim.v' to AST representation.
Generating RTLIL representation for module `\LUT1'.
Generating RTLIL representation for module `\LUT2'.
Generating RTLIL representation for module `\LUT3'.
Generating RTLIL representation for module `\LUT4'.

*** REMOVED A BUNCH OF LOGS ***

2.30. Printing statistics.

=== top ===

   Number of wires:                130
   Number of wire bits:            177
   Number of public wires:         130
   Number of public wire bits:     177
   Number of ports:                  2
   Number of port bits:              7
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                114
     ALU                            54
     DFFE                            6
     DFFR                           24
     GND                             1
     IBUF                            1
     LUT1                           12
     LUT2                            1
     LUT4                            3
     MUX2_LUT5                       4
     MUX2_LUT6                       1
     OBUF                            6
     VCC                             1

2.31. Executing CHECK pass (checking for obvious problems).
Checking module top...
Found and reported 0 problems.

2.32. Executing JSON backend.

End of script. Logfile hash: 691ee5878c, CPU: user 0.22s system 0.01s
Yosys 0.40 (git sha1 a1bb0255d65, clang++ 15.0.0 -fPIC -Os)
Time spent: 28% 20x read_verilog (0 sec), 11% 1x abc9_exe (0 sec), ...
```

What am I doing wrong?

alaindargelas, 2024-05-01:
Do not build nor install Yosys separately.
Simply build Synlig, it will build Yosys as a part of the synlig build and other components and it will all work together.
Use the cmake build instructions:
https://github.com/chipsalliance/synlig?tab=readme-ov-file#embedding-synlig-in-a-larger-cmake-based-project
make -f cmake-makefile

zaun, 2024-05-02:
ahh, that explains it, I have yosys installed already. Am I correct in saying that systemverilog is not a plugin for yosys then? its a "plugin" but requires a modified yosys or it won't work? If yosys does need modified, why not put a patch into the main yosys package?

I'll have to find time to uninstall the system yosys and try again. at the moment I have things sortof working with sv2v.

alaindargelas, 2024-05-02:
Correct, it does not really work as a plugin as it is, you have to build it alltogether the way it was setup.

The Synlig plugin is not compatible with the latest Yosys api, so you can't use the latest yosys starting about 2 months ago. Synlig checks out the compatible version and patches it.

No funds to integrate properly with Yosys are available at this point (both on Yosys and Synlig side).

There is another integration model used here were both are built separately, but it is an expert build level that I would not recommend unless you want to tinker with cmake:

https://github.com/os-fpga/Raptor_Tools/tree/8705ea7ef18c3b0f6aa41e490f1edb03a672f08e/parser_plugins