chipsalliance/synlig #2432: Mismatch between yosys-synlig and bsg_sv2v

dpetrisko, 2024-05-21:
Hi, I am trying to synthesis a systemverilog design with synlig (https://github.com/dpetrisko/tt07-dll/tree/main). With the plugin installed from main, the design is able to parse and synthesize to completion. Unfortunately, the post-synthesis simulation fails. However, using bsg_sv2v (https://github.com/bespoke-silicon-group/bsg_sv2v) to write out verilog-2005 then having yosys synthesize that works great. So there seems to be a mismatch in parsing / the plugin

What would be helpful for submitting a test case? Do you have an example regression test case I can look at? Unfortunately I'm not familiar with the open formal tools so it might be difficult for me to produce anything other than netlists

Thanks!

tgorochowik, 2024-05-22:
Hi, our test suite is here: https://github.com/chipsalliance/synlig/tree/main/tests so in general it would be best to minimize the test around the part that fails in your post-synthesis simulation.

Before you do that though, Synling currently does not work correctly with mainline yosys, it's being tested with the one specified as a submodule here: https://github.com/chipsalliance/synlig/tree/main/third_party (so ea7818d31bb2533d4ecceb2ed1bcf4a22b850453 at this moment).
I think it would make sense for you to check this version first, it's possible something changed in yosys since then that causes the issues you see

dpetrisko, 2024-05-22:
Okay thanks! I am using ea7818d31bb2533d4ecceb2ed1bcf4a22b850453, so looks like it's a modern problem. I will work on the test case and get back to you!

dpetrisko, 2024-06-04:
Closing this, will open a PR with the test case!

closed 2024-06-04 by dpetrisko