chipsalliance/synlig #2505: Synlig has a problem with nested if else

Kreijstal, 2024-08-04:
Given this code.

```
  95   │   always_ff @(posedge clk_i, negedge rst_ni) begin
  96   │     if (~rst_ni) begin
  97   │       isdifferent <= '0;
  98   │       middleVariable <= '0;
  99   │     end else begin
 100   │       if (middleVariable != regC) begin
 101   │         isdifferent <= 1'b1;
 102   │         middleVariable <= regC;
 103   │       end // if(middleVariable)
 104   │          else if (isdifferent==1'b1&&(regB!=cnt||shift!=5'd7))
 105   │          begin
 106   │         isdifferent <= 1'b1;
 107   │       end else begin
 108   │         isdifferent <= 0'b0;
 109   │       end
 110   │     end  // if (~rst_ni) else
 111   │   end
```

with error message
```
[SNT:PA0207] /home/kreijstal/git/riscvxu/src/rtl/student/student_rlight.sv:99:8: Syntax error: rule conditional_statement failed predicate: _input->LA(1) != ELSE?,
    end else begin
        ^-- /home/kreijstal/git/riscvxu/src/rtl/student/slpp_all/lib/work/student_rlight.sv:99:8:.
[SNT:PA0207] /home/kreijstal/git/riscvxu/src/rtl/student/student_rlight.sv:107:10: Syntax error: rule conditional_statement failed predicate: _input->LA(1) != ELSE?,
      end else begin
          ^-- /home/kreijstal/git/riscvxu/src/rtl/student/slpp_all/lib/work/student_rlight.sv:107:10:.
[SNT:PA0207] /home/kreijstal/git/riscvxu/src/rtl/student/student_rlight.sv:108:24: Syntax error: extraneous input ''b0' expecting ';',
        isdifferent <= 0'b0;
                        ^-- /home/kreijstal/git/riscvxu/src/rtl/student/slpp_all/lib/work/student_rlight.sv:108:24:.
                        
```
synlig does not like this type of structure.

Kreijstal, 2024-08-04:
is this a UHDM only bug, and should be filed there?

alaindargelas, 2024-08-05:
Your best bet is to rewrite this code in a less convoluted way.
But your can check if Vivado supports the code.
Provide a fully compiling testcase with all declarations, not just a code snippet and file the issue under Surelog.

Kreijstal, 2024-08-05:
Vivado does simulate synthetise and tested in hardware in fpga.