chipsalliance/synlig #2547: failure while running `make build`

jeras, 2024-09-09:
The master head fails to build (with `make build`) with error:
```sh
cp: cannot stat 'techlibs/gatemate/lut_tree_cells.genlib': No such file or directory
make[1]: *** [/home/izi/VLSI/synlig/third_party/yosys/techlibs/gatemate/Makefile.inc:30: share/gatemate/lut_tree_cells.genlib] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/izi/VLSI/synlig/build/release/synlig'
make: *** [Makefile:472: synlig] Error 2
```

tgorochowik, 2024-09-10:
Could you please provide more context? What are you executing and in which directory? Are all of the submodules fetched correctly?

Please try on ma fresh clone and paste all the commands and their outputs here, so we can help you

jeras, 2024-09-10:
From a clean clone, and following instructions in the README, **I got a working build**. The steps I performed:
```
git clone git@github.com:chipsalliance/synlig.git
sudo apt install -y gcc-11 g++-11 build-essential cmake tclsh ant default-jre swig google-perftools libgoogle-perftools-dev python3 python3-dev python3-pip uuid uuid-dev tcl-dev flex libfl-dev git pkg-config libreadline-dev bison libffi-dev wget python3-orderedmultidict
cd synlig/
git submodule sync
git submodule update --init --recursive third_party/{surelog,yosys}
make install -j$(nproc)
```

I will try to clean up my original clone and try there again.

closed 2024-10-13 by jeras