chipsalliance/synlig #2560: Recent builds cannot find share/

thirtythreeforty, 2024-09-12:
After downloading and unpacking the [most recent release as of this writing](https://github.com/chipsalliance/synlig/releases/tag/2024-09-12-aa98f61), I get an error when attempting synthesis:

```
~/synlig-new$ tar xf synlig-aa98f61-2024-09-12.tar.gz
~/synlig-new$ usr/local/bin/synlig

yosys> read_systemverilog ../yosys/top.sv

1. Executing Verilog with UHDM frontend.
[INF:CM0023] Creating log file "/home/ubuntu/synlig-new/slpp_all/surelog.log".
[INF:CP0300] Compilation...
[INF:CP0303] /home/ubuntu/yosys/top.sv:1:1: Compile module "work@top".
[INF:CP0302] Compile class "work@mailbox".
[INF:CP0302] Compile class "work@process".
[INF:CP0302] Compile class "work@semaphore".
[NTE:CP0309] /home/ubuntu/yosys/top.sv:3:16: Implicit port type (wire) for "led".
[INF:EL0526] Design Elaboration...
[NTE:EL0503] /home/ubuntu/yosys/top.sv:1:1: Top level module "work@top".
[WRN:EL0500] /home/ubuntu/yosys/top.sv:9:3: Cannot find a module definition for "work@top::BUFG".
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 2.
[NTE:EL0510] Nb instances: 2.
[NTE:EL0511] Nb leaf instances: 1.
[WRN:EL0512] Nb undefined modules: 1.
[WRN:EL0513] Nb undefined instances: 1.
[INF:UH0706] Creating UHDM Model...
[INF:UH0707] Elaborating UHDM...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 3
[   NOTE] : 6
Generating RTLIL representation for module `\top'.

yosys> synth_xilinx

2. Executing SYNTH_XILINX pass.
ERROR: init_share_dirname: unable to determine share/ directory!
```

This problem does not occur with the prior release `synlig-967ce78-2024-09-07.tar.gz`.

kamilrakoczy, 2024-09-12:
Could you please check if problem is still present in [current most recent release](https://github.com/chipsalliance/synlig/releases/tag/2024-09-12-ca23376)?

We just merged fix that changes release package and now there shouldn't be any problems with finding shared directory.
I just checked it and this release works for me.

thirtythreeforty, 2024-09-12:
Looks like it's working for me as well. Thank you!

closed 2024-09-12 by thirtythreeforty