chipsalliance/synlig #2677: Does synlig plugin attempt to load `write_ilang` ?

hzeller, 2024-11-26:
When attempting to load the synlig plugin, yosys errors out complaining that `write_ilang` pass already exists, as if the synlig plugin recusively attempts to load that other plugin ?

```
yosys> plugin -i path/to/systemverilog.so
Warning: Using synlig as yosys plugin is deprecated. It is recommended to build synlig as standalone binary.
ERROR: Unable to register pass 'write_ilang', pass already exists!
```

kamilrakoczy, 2024-11-28:
Yosys recently dropped `read/write_ilang` alias: https://github.com/YosysHQ/yosys/pull/4704

We added this alias back on synlig side when we were updating yosys version: https://github.com/chipsalliance/synlig/pull/2668.

In https://github.com/chipsalliance/synlig/pull/2685 we added check if this pass is already registered and only register this alias if it isn't.

We encourage users to use synlig as as standalone binary to make sure we don't see such issues.

closed 2024-11-28 by kamilrakoczy

hzeller, 2024-12-03:
Thanks.
Right now I need the plug-in feature as I am maintaining the NixOS package, and having a smoothly working plugin that works with an existing Yosys is preferrable from the packaging point-of-view than pulling in a vendored yosys.

PR #2685 (2024-11-28): Add alias for write_rtlil only when ilang frontend doesn't exist
Fixes: #2677
commit b65b2868 (2024-11-28): Add alias for write_rtlil only when ilang frontend doesn't exist (#2685)

Fixes: #2677