chipsalliance/synlig #2717: read_systemverilog generates bad netlist for simple module

kcameron4456, 2024-12-24:
I'm working on a hobbyist product to implement an Othello game lookahead solver on an FPGA. My code uses SystemVerilog constructs so I need a compatible synthesis tool. I found synlig a few days ago and have been attempting run my code through it. While wading through lots of issues (mostly on my side), I found this.

I'm using the "main" branch of synlig I cloned and built on 12/22/24:
```
$ git branch
* main
$ synlig --version
Synlig (git sha1 2d838edc9, c++ 14.2.1 -fPIC -O3)
```

Shortened (non-working) Module (FindMoveMask_sz3.v):
```
// Generated by GenFindMoveMask.pl
//
// This module generates a mask of all possible moves for black from the given board (Blk/Wht)
//

`define BOARD [8:0]

module FindMoveMask (
    Blk,
    Wht,
    Moves
);

input  `BOARD Blk;
input  `BOARD Wht;
output `BOARD Moves;

// moves only allowed to empty squares
wire `BOARD Emp = ~(Blk | Wht);

// Masks of moves considering the state in each of the 8 possible directions
wire `BOARD Move_N;
wire `BOARD Move_E;
wire `BOARD Move_W;
wire `BOARD Move_S;

// Combine all moves
wire `BOARD Moves = Move_N | Move_E | Move_W | Move_S ;

// Create all moves
assign Move_N  [00] = 1'b0;
assign Move_N  [01] = 1'b0;
assign Move_N  [02] = 1'b0;
assign Move_N  [03] = 1'b0;
assign Move_N  [04] = 1'b0;
assign Move_N  [05] = 1'b0;
assign Move_N  [06] = Emp[06] && Wht[03] && Blk[00];
assign Move_N  [07] = Emp[07] && Wht[04] && Blk[01];
assign Move_N  [08] = Emp[08] && Wht[05] && Blk[02];

assign Move_E  [00] = 1'b0;
assign Move_E  [01] = 1'b0;
assign Move_E  [02] = Emp[02] && Wht[01] && Blk[00];
assign Move_E  [03] = 1'b0;
assign Move_E  [04] = 1'b0;
assign Move_E  [05] = Emp[05] && Wht[04] && Blk[03];
assign Move_E  [06] = 1'b0;
assign Move_E  [07] = 1'b0;
assign Move_E  [08] = Emp[08] && Wht[07] && Blk[06];

assign Move_W  [00] = Emp[00] && Wht[01] && Blk[02];
assign Move_W  [01] = 1'b0;
assign Move_W  [02] = 1'b0;
assign Move_W  [03] = Emp[03] && Wht[04] && Blk[05];
assign Move_W  [04] = 1'b0;
assign Move_W  [05] = 1'b0;
assign Move_W  [06] = Emp[06] && Wht[07] && Blk[08];
assign Move_W  [07] = 1'b0;
assign Move_W  [08] = 1'b0;

assign Move_S  [00] = Emp[00] && Wht[03] && Blk[06];
assign Move_S  [01] = Emp[01] && Wht[04] && Blk[07];
assign Move_S  [02] = Emp[02] && Wht[05] && Blk[08];
assign Move_S  [03] = 1'b0;
assign Move_S  [04] = 1'b0;
assign Move_S  [05] = 1'b0;
assign Move_S  [06] = 1'b0;
assign Move_S  [07] = 1'b0;
assign Move_S  [08] = 1'b0;

endmodule
```

I read the source  with this:
```
synlig> read_systemverilog FindMoveMask_sz3.v 
...
[INF:UH0707] Elaborating UHDM...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 5
Generating RTLIL representation for module `\FindMoveMask'.
```

I synthesize thusly:
```
synlig> synth_xilinx
...
=== FindMoveMask ===

   Number of wires:                  9
   Number of wire bits:             81
   Number of public wires:           7
   Number of public wire bits:      63
   Number of ports:                  3
   Number of port bits:             27
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 27
     IBUF                           18
     OBUF                            9

   Estimated number of LCs:          0
```

Note that only buffers are generated. When I write the verilog, I see only buffers in the netlist and weird assignments:
```
synlig> write_verilog Test.sv
...
Dumping module `\FindMoveMask'.
synlig> exit
$ tail Test.sv 
  (* keep = 32'd1 *)
  IBUF _28_ (
    .I(Wht[8]),
    .O(_01_[8])
  );
  assign Move_E = 9'bx00x00x00;
  assign Move_N = 9'bxxx000000;
  assign Move_S = 9'b000000xxx;
  assign Move_W = 9'b00x00x00x;
endmodule
```

If I use read_verilog, I get more reasonable results:
```
synlig> read_verilog FindMoveMask_sz3.v 
...
Successfully finished Verilog frontend.
synlig> synth_xilinx 
...
=== FindMoveMask ===

   Number of wires:                 19
   Number of wire bits:            104
   Number of public wires:           7
   Number of public wire bits:      55
   Number of ports:                  3
   Number of port bits:             27
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 35
     IBUF                           18
     LUT4                            4
     LUT6                            4
     OBUF                            9

   Estimated number of LCs:          8
synlig> exit
$ tail Test.sv2 
  assign _08_[3:1] = { _06_[2], _07_[1], _07_[3] };
  assign { _09_[4], _09_[2:0] } = { _05_[1], _08_[0], _05_[4], _05_[0] };
  assign { _07_[2], _07_[0] } = { _06_[2], _05_[2] };
  assign _10_[4:0] = { _05_[3], _05_[4], _05_[2], _06_[1], _09_[3] };
  assign _11_[4:0] = { _09_[3], _06_[1], _05_[3], _05_[1], _08_[0] };
  assign { _04_[8:5], _04_[3:0] } = { _05_[1], _06_[3], _05_[4], _07_[3], _07_[1], _09_[3], _06_[0], _05_[3] };
  assign { Move_E[7:6], Move_E[4:3], Move_E[1:0] } = 6'h00;
  assign Move_N[5:0] = 6'h00;
  assign { Move_W[5:4], Move_W[2:1] } = 4'h0;
endmodule
```

alaindargelas, 2025-07-28:
The fix is pushed here: https://github.com/alainmarcel/uhdm2rtlil

closed 2025-07-28 by kcameron4456

alaindargelas, 2025-07-28:
My fix Is not entirely correct. It produces assign statements for ports initial values. I'll have to refine it