chipsalliance/synlig #2769: Break doesn't compile within a case

mphipps1, 2025-07-17:
break doesn't compile within a case statement. 
Test case:
module test ( input logic clock);
    logic [2:0] state, random;
    always_ff @(posedge clock) begin
        for (int i=0;i<16;i++) begin // do it a bunch of times...
            unique case (state)
                0: if (random == 0) break;
                1: if (random == 1) break;
            endcase
        end // for
    end // always
endmodule

Error Message (only the relevant parts included):
Generating RTLIL representation for module `\test'.
ERROR: Abort in /root/synlig/synlig/third_party/yosys/frontends/ast/genrtlil.cc:871.

mphipps1, 2025-07-28:
It doesn't compile for me:
$ synlig --version
Synlig (git sha1 2d838edc9, g++ 12.2.0-14 -fPIC -O3)

$ synlig 

synlig> read_systemverilog test.sv

1. Executing SystemVerilog frontend.
[INF:CM0023] Creating log file "[mypath]slpp_all/surelog.log".
[INF:CP0300] Compilation...
[INF:CP0303][mypath]test.sv:1:1: Compile module "work@test".
[INF:CP0302] Compile class "work@mailbox".
[INF:CP0302] Compile class "work@process".
[INF:CP0302] Compile class "work@semaphore".
[INF:EL0526] Design Elaboration...
[NTE:EL0503] [mypath]test.sv:1:1: Top level module "work@test".
[NTE:EL0508] Nb Top level modules: 1.
[NTE:EL0509] Max instance depth: 1.
[NTE:EL0510] Nb instances: 1.
[NTE:EL0511] Nb leaf instances: 0.
[INF:UH0706] Creating UHDM Model...
[INF:UH0707] Elaborating UHDM...
[  FATAL] : 0
[ SYNTAX] : 0
[  ERROR] : 0
[WARNING] : 0
[   NOTE] : 5
Warning: [mypath]test.sv:4: Post-incrementation operations are handled as pre-incrementation.
Generating RTLIL representation for module `\test'.
ERROR: Abort in /root/synlig/synlig/third_party/yosys/frontends/ast/genrtlil.cc:871.

------------------

After building uhdm2rtlil, it does compile my test.sv to a Surelog.uhdm, but then YOSYS crashes on the output:

/----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.51 (git sha1 c4b519022, clang++ 19.1.1 -fPIC -O3)

-- Running command `plugin -i ../build/uhdm2rtlil.so; read_uhdm ./surelog.uhdm; synth' --

1. Executing UHDM frontend.
Reading UHDM file: ./surelog.uhdm
UHDM Elaboration...
Module: work@test () Flat:1, Top:0
Module: work@test () Flat:1, Top:0
Module: work@test (work@test) Flat:0, Top:1
Module: work@test (work@test) Flat:0, Top:1
UHDM: Starting import_design
UHDM: Identified top-level module: test
UHDM: Found 1 packages in design
UHDM: About to import package: builtin
UHDM: Importing package: builtin
UHDM: Finished importing package builtin
UHDM: Found 1 module definitions in AllModules
UHDM: Skipping top module test from AllModules (will import from hierarchy)
UHDM: Found 1 top modules in design
UHDM: Importing top module hierarchy starting from: work@test
UHDM: No param_assigns found for test
UHDM: Final param_signature: test
UHDM: Importing module from hierarchy: test
UHDM: Processing module, VpiDefName() returns: 'work@test', VpiName() returns: 'work@test'
UHDM: Stripped work@ prefix, using: 'test'
UHDM: Found 1 ports to import
UHDM: About to import port: 'clock'
UHDM: get_width analyzing object type
UHDM: Found port object
UHDM: Port has typespec, calling get_width_from_typespec
UHDM: Analyzing typespec for width determination
UHDM: Typespec UhdmType = 2304
UHDM: Found ref_typespec, following reference
UHDM: Following to actual typespec (UhdmType = logic_typespec)
UHDM: Analyzing typespec for width determination
UHDM: Typespec UhdmType = 2212
UHDM: ExprEval returned size=1 for typespec
UHDM: Creating wire 'clock' (width=1, mode_keep_names=false)
UHDM: About to call module->addWire('\clock', 1)
UHDM: Successfully created wire '\clock'
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
/----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.51 (git sha1 c4b519022, clang++ 19.1.1 -fPIC -O3)

-- Running command `plugin -i ../build/uhdm2rtlil.so; read_uhdm ./surelog.uhdm; synth' --

1. Executing UHDM frontend.
Reading UHDM file: ./surelog.uhdm
UHDM Elaboration...
Module: work@test () Flat:1, Top:0
Module: work@test () Flat:1, Top:0
Module: work@test (work@test) Flat:0, Top:1
Module: work@test (work@test) Flat:0, Top:1
UHDM: Starting import_design
UHDM: Identified top-level module: test
UHDM: Found 1 packages in design
UHDM: About to import package: builtin
UHDM: Importing package: builtin
UHDM: Finished importing package builtin
UHDM: Found 1 module definitions in AllModules
UHDM: Skipping top module test from AllModules (will import from hierarchy)
UHDM: Found 1 top modules in design
UHDM: Importing top module hierarchy starting from: work@test
UHDM: No param_assigns found for test
UHDM: Final param_signature: test
UHDM: Importing module from hierarchy: test
UHDM: Processing module, VpiDefName() returns: 'work@test', VpiName() returns: 'work@test'
UHDM: Stripped work@ prefix, using: 'test'
UHDM: Found 1 ports to import
UHDM: About to import port: 'clock'
UHDM: get_width analyzing object type
UHDM: Found port object
UHDM: Port has typespec, calling get_width_from_typespec
UHDM: Analyzing typespec for width determination
UHDM: Typespec UhdmType = 2304
UHDM: Found ref_typespec, following reference
UHDM: Following to actual typespec (UhdmType = logic_typespec)
UHDM: Analyzing typespec for width determination
UHDM: Typespec UhdmType = 2212
UHDM: ExprEval returned size=1 for typespec
UHDM: Creating wire 'clock' (width=1, mode_keep_names=false)
UHDM: About to call module->addWire('\clock', 1)
UHDM: Successfully created wire '\clock'
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)

mphipps1, 2025-07-28:
I am sorry - I don't follow. The test case above is a cut down version of several hundred lines of code, but it is a complete Verilog example that does crash the compiler. You said that it creates an empty module; it doesn't do that here - the compiler crashes with an ERROR: abort. Does it not do that for you?

Attaching the test file.

[test.sv.txt](https://github.com/user-attachments/files/21476298/test.sv.txt)

alaindargelas, 2025-07-30:
Synlig has 3 components: Surelog, UHDM and the code in this repo (Synlig).
Surelog and UHDM are maintained. This repo is not maintained anymore.
The only fixes I can make are in Surelog and UHDM, and for some issues, that is enough, others, nothing can be done. This repo uses the Yosys AST API as a target which is a moving target (Yosys changes) so there is also the problem of keeping up with Yosys and backward compatibiliy .
uhdm2rtl is a 1 week old project, it is 99% vibe coded, it targets the Yosys RTLIL API which is stable. Also since it is vibe coded, it will be easy to maintain and expand. It does not yet have the same breath as synlig, but the goal is that in a couple of months it should match it.